MEGA65/mega65-tools #2: PRG load after ROM load fails

Taygair, 2020-09-06:
After loading a ROM file by m65 any loaded PRG file (maybe others too) on loading:

[T+0sec] Detecting C64/C65 mode status.
[T+0sec] Could not determine C64/C65/MEGA65 mode.
[T+0sec] Loading file '/.../.../Turbo Disk 65 v3.prg'
[T+0sec] Could not determine C64/C65/MEGA65 mode.
ERROR: Should be in C65 mode, but don't seem to be.

I looks like m65 has a problem to evaluate the MEGA65 mode after loading a ROM file. 

Loading a PRG file with RESET option works but that of course removes the ROM file.

Taygair, 2021-04-11:
Might be related to this case which belongs here: https://github.com/MEGA65/m65connect/issues/6

gurcei, 2021-04-25:
I've confirmed that Paul's recent changes have fixed this issue.

closed 2021-04-25 by gurcei