MEGA65/mega65-tools #22: MEGA65_FTP: Make 'del' command work on directories too

gurcei, 2021-01-27:
Presently, I suspect 'del' command only works on files. Tayger recently put on my radar the desire to delete directories too, which I don't think the current implementation considers well.

Will probably need to devise a recursive algorithm that dives into all sub-directories too and deletes all files within there also.

gurcei, 2021-02-03:
- rdpeake mentioned that 'del' command was missing from the help menu, so I can add it in on this card too.

gurcei, 2022-04-12:
Presently, the code was deleting a dir by just removing the dir-entry for the dir.

This isn't ideal, as it means any child files or child dirs within this dir will be orphaned.

It'd be better to implement a recursive strategy to delete all files and sub-folders first, then delete the specified dir.

Also, would be courteous to warn the user first "This dir contains xxx dirs and xxx files, are you sure you want to delete?"

Taygair, 2023-05-12:
From M65Connect perspective it would be totally ok if mega65_ftp refuses DIR deletion if not empty. So the user has to delete all its content first before able to delete the directory. In a further version I can implement the recursive stuff on the M65Connect side. 

To cover the CRUD philosophy: **Renaming** a directory name would also be nice (if not already implemented)

gurcei, 2023-05-12:
Hopefully I can get recursive deletion of files within a directory working one day, and save M65Connect the trouble of doing it.

I 'think' renaming of directories works already.

Taygair, 2023-05-12:
No hurry at all on recursive deletion! It would already be just great if deletion of an empty directory is possible (deletion refused if not empty).
Great will test the renaming later on.

nobruinfo, 2024-11-08:
@gurcei @Taygair Consider this ticket now in the works with me.

nobruinfo, 2024-11-09:
Forgot to post here. Up for broader testing: https://github.com/MEGA65/mega65-tools/pull/207

gurcei, 2024-11-09:
Thanks for putting time into making this happen @nobruinfo :) I've approved and merged in your pull request on this, but will still keep the ticket open in relation to other aspects which I mentioned in the PR comment, which I'll paste here too.

I'm happy with the approach you've taken with assessing sub-directories and files within the folder to be deleted.

I do worry a little that the implementation only delete the main directory and not the files/sub-folders within, and it means that these items and their fat file entries will be left dangling on the sdcard as orphans, and in time, it'd be better to recursively go through all sub-folders and files and delete them one by one too.

But anyways, for this first iteration, I can make peace with that, though let's leave the ticket open as a reminder that we can come back to this matter later on to refine it further.

Another thing that would be nice to add over time is a few tests into this file relating to the new delete behaviour, especially the scenario you described where you bumped into some misbehaviour:

https://github.com/MEGA65/mega65-tools/blob/development/gtest/mega65_ftp_test.cpp

It doesn't have to be done for this iteration though, can save that for another time too :)

I'll mention this comment on the ticket too, for future reference.

gurcei, 2024-11-09:
No probs, and yep, I'm in no hurry for 1, as most folks have pretty large sd-cards, and a few orphaned files here and there shouldn't cause them too much grief.

For 2, its ok, there's no rush there either, when time permits, or even ok if you prefer me dive into it at some later stage.

I don't always go the extra mile with writing tests myself, but found the urge to do so greater when it came to mega65_ftp, as in the bad old days of the tool, it had many bugs that would trash the entire sd-card and you'd have to start all over :)

So with this new functionality, it'd also be comforting to add some tests one day, just to give an extra sense of confidence that we've locked in some working functionality :)

nobruinfo, 2024-11-10:
@gurcei By numbers:
1. I think I should add some more stats to the `dir` so we all could see how much free space there still is and whatever else could be of help. I'm not too sure the FAT is getting fuller. I'd like to convince myself. ;)
2. Don't take me wrong, I'm also full forward for testing. I only meant to build up some time and motivation for it. A fun thing found just now is the `mkdir` seems to be the pack of fleas, long before deleting. I have my batch file populating my Xemu image file, also across ethernet and both look really funny. I very much expect my previous `del` recoveries to be `mkdir` inflicted and plan to investigate the latter first.

nobruinfo, 2024-11-11:
@gurcei You of course were right. I added the number of remaining clusters to the `dir` output. Executing a few `mkdir`, `put` and later `del` to the whole structure does **not** liberate clusters back to their original number of free. All these improvements will be part a pull request to come.
The hard part will now be to create the recursive routine with all the global variable trickery going on withing mega65_ftp. So will need a little.
I also see you do your things but I am unfortunately not being able to follow completely what you're documenting in Discord. Also there still is one of your videos I haven't watched yet. Still lots for me to explore.

nobruinfo, 2024-11-16:
@gurcei I'd say this one test I've just added pretty much verifies across level dir structure deletion is basically working. Should there still be things to add I'd be happy to. Besides that I think this one feature could now be called complete.

nobruinfo, 2026-10-21:
@gurcei When you merged my pull requests last year the branches were deleted. One of them belonged to this ticket. I expect this to now work. Please retest and if fine, please close the ticket. Thank you.