MEGA65/mega65-tools #23: MEGA65_FTP: make 'put' work for directories

gurcei, 2021-01-27:
Presently, 'put' only works for single files.

Will need to add some recursive logic in order for it to recurse into subdirectories and upload each files within

nobruinfo, 2025-07-18:
That's not all there is to it. `put *` allows uploading a series of files already. Maybe this wasn't the case back when this ticket was opened.

I'd suggest to make a first approach implementing this by checking the to be uploaded for existance on the remote machine or Xemu image. If so I'd like the transfer prematurely. Were we to try this differently the process would have to handle single existing files or folders leaving what's here and there utterly incomprehensible. Also the switch `-y` would make it hard to handle for tools like M65Connect.

nobruinfo, 2025-07-19:
Testing nested dir structures and handling files with relative paths towards the destination reveals a huge bag of fleas. Using `/` to access certain levels must have had its culprits for possible quite some time:

```cmd
MEGA65 SD-Card:/MEGA65> dir MEGA65
       <DIR> | 2025-07-19 09:23:50  | .            |
       <DIR> | 2025-07-19 09:23:50  | ..           |
       <DIR> | 2025-07-19 09:59:26  | MEGA65       |
3 Dir(s), 0 File(s)
MEGA65 SD-Card:/MEGA65> dir MEGA65 MEGA65/MEGA65
       <DIR> | 2025-07-19 09:23:50  | .            |
       <DIR> | 2025-07-19 09:23:50  | ..           |
       <DIR> | 2025-07-19 09:59:26  | MEGA65       |
3 Dir(s), 0 File(s)
MEGA65 SD-Card:/MEGA65> dir MEGA65 MEGA65/MEGA65/MEGA65
       <DIR> | 2025-07-19 09:23:50  | .            |
       <DIR> | 2025-07-19 09:23:50  | ..           |
       <DIR> | 2025-07-19 09:59:26  | MEGA65       |
3 Dir(s), 0 File(s)
```

Also it seems entering unknown commands no longer is throwing an error:

```cmd
MEGA65 SD-Card:/> jhasghjagdhjgf
MEGA65 SD-Card:/>
```

Maybe I should stop handing in pull requests now and write new tickets first to avoid showstoppers. :/

nobruinfo, 2026-10-21:
@gurcei When you merged my pull requests last year the branches were deleted. One of them belonged to this ticket. I expect this to now work. Please retest and if fine, please close the ticket. Thank you.