MEGA65/mega65-tools #40: M65: Injecting a basic prg with m65 not working properly

hstampfl, 2021-02-13:
Injecting a BASIC V10 prg with m65 tool does not get correctly injected.
The program get transfered and it correctly starts, but editing the program crashes the mega65.

This issue was not seen before because an assembly program mostly won't be edited after executing.

See more details on discord:
https://discord.com/channels/719326990221574164/795760112425369610/809155791718187060

Use following steps to reproduce:
1. unzip and transfer attached program with m65 (on windows: m65 -l COMxxx -F -r rasterwait.prg)
2. Press RUN/STOP
3. Execute LIST
4. Edit last Line and press RETURN
5. You should see an similar screen as attached screenshot.
6. 
[image]

[rasterwait.zip](https://github.com/MEGA65/mega65-tools/files/5975957/rasterwait.zip)

Edilbert, 2021-02-15:
After transferring the program, the pointer "text_top" ($82/$83) must be set to the end of the program text.
It is the same value, like the (eal/eah) = End Address Load = ($ae/$af).
The editor needs this information, if not set to the correct value, the editor will produce strange results or crash.

gardners, 2021-04-25:
In _theory_ this should be correctly handled by having the X and Y registers loaded with the top address of the LOAD.
But now that we trap the LOAD vector in a more generic manner (see commit for fixing issue #2 ), we probably need to set this directly.

gardners, 2021-04-25:
Seems to be fixed by the above.

gurcei, 2021-04-25:
I've confirmed that this has resolved the issue.

closed 2021-04-25 by gurcei