MEGA65/mega65-tools #43: monitor_save & monitor_load don't build for Windows.

JettMonstersGoBoom, 2021-02-17:
with mingw64. initially fails with termios.h missing.

gurcei, 2021-02-20:
Could you share a direct copy/paste of the error message here (just to get the details of which file contained the include).

Aslo a note from my experiences, with mingw64 not every tool within the suite is guaranteed to build at this stage.

So personally, when I'm on mingw64 in windows, I aim to build windows specific targets that are known to work, like

- make bin/mega65_ftp.exe
- make bin/m65.exe
- make bin/bit2core.exe

gurcei, 2021-02-20:
I also think these two are older tools in the suite. There was a card (#31) hoping to update them to make use of the m65common library functions, this might help matters on this card too

JettMonstersGoBoom, 2021-02-22:
I posted an issue at @gardners request, after he mentioned I could try Monitor_save/load for debugging an issue. seems a chunk of these tools are not building with Mingw .

gurcei, 2022-05-05:
Hmm, in a past discord convo with @gardners, I recall him mentioning that the monitor_load functionality has already migrated into the m65 tool, so perhaps it's time to discard it.

As for monitor_save, perhaps it is another tool that can have its functionality migrate into the m65 tool too.

closed 2022-07-30 by lydon42