MEGA65/mega65-tools #50: M65 Add softreset option

jesperrula, 2021-03-16:
It would be nice if there was an option to do a softreset instead of a full core reset when sending a file to the Mega65. It could save a lot of waiting time each time we compile and run a prg. 
Any other ideas for making the push of PRG's to the Mega65 faster would be very welcome too.

gurcei, 2021-07-16:
Hmm, I recently bumped into the `SYS 58552` call to perform a soft-reset from C64 back to C65-mode.

I'm wondering if there's any other SYS-call to perform the soft-reset while still in C65 mode?

If so, you could get m65 to type such a command for you, with something like `m65 -l COM7 -T "sys 58552"`

gurcei, 2021-07-16:
TheChief in discord made the following suggestions:
- https://discord.com/channels/719326990221574164/795760112425369610/865422805897969674

> Please try $F9A7 / 63911 @C65 mode, like $FCE2 / 64738 @C64 mode, this is where the address $FFFC/FFFD for the reset / power on vector points to

gurcei, 2021-07-16:
Just to confirm, were you presently using the '-F' argument and is this the reset behaviour you found too slow?

Just wanted to confirm this first, in-case you weren't using it, and perhaps you could give it a try and see if it suited your needs.

gurcei, 2021-07-16:
Tayger has told me that the -F argument resets things quite fast already. Given that, I think it's ok for me to close this ticket.

closed 2021-07-16 by gurcei