MEGA65/mega65-tools #52: MEGA65_FTP: Add a 'defrag' command

gurcei, 2021-04-11:
A little Discord discussion on this recently:
- https://discord.com/channels/719326990221574164/805492126128930846/830945137701290004

My thoughts for it are:

1) For a future defrag MYFILE.D81 command, behind the scenes, I'd just be doing a get, delete and put behind the scenes.

2) For the case of trying to put MYFILE.D81 where it already exists on the sd-card in a fragmented state, I've added a function into mega65_ftp.c called is_fragmented(filename) to check if an existing file is fragmented. So the logic here could be:
On upload_file()
{
  If file exists on sd-card then
  {
    If `is_fragmented(filename)` then
      delete existing file
  }
  copy across the new file
}

lydon42, 2022-09-25:
Implemented in:
52a70474
c98499b1
9901cc80

is_fragmented can still be optimized a bit (no need to go through the whole file, if it finds the first non-consecutive cluster, it can return fragmentation directly). And it needs to be used, which is not done yet.

lydon42, 2024-05-20:
So this is no defrag command, but it will not reuse an existing files clusters if it was fragmented, and instead delete it, so that it is recreated in one continues clusterchain.

I think that should solve this issue.

nobruinfo, 2026-10-21:
@lydon42 @gurcei This works very well and now can be further enhanced in two ways:
 
1. @Taygair suggested not to alter the existing `dir` but to add a `direx` command. A later merge of these would be no fuss anyways. The new command adds an extra column (or columns to come) but is not used by M65Connect™ currently, hence not breaking compatibilty. In the extra column the fragmentation status is shown taken from the function discussed above. I have this active in my test fork for months and it's showing a fragmented file from time to time. There's also known ways to `del` files leaving them as corrupted. I intend to leave this until after the routine here is tested in all expansions.
2. Similar to the recently established `df` command we could invent a `fragreport` command to recursively visit all directories of the storage to only list fragmented files. Again this will be very slow and people will have to [Ctrl][Break] out of mega65_ftp entirely to stop this. Fragmented files listed seem to include corrupted it seems. So I'll print a hint to this with the screen output.

May I therefore suggest to leave this ticket open for me to hand in a pull request for you to review? One or the other might be easily removed from the PR if the general strategy with this tool suite won't allow for such ideas.