MEGA65/mega65-tools #88: M65: Sending prg file doesn't work anymore

Taygair, 2021-08-19:
Majikeyric reported this one in "closed-roms":

With the 920217 ROM m65 doesn't work anymore to inject PRG (920215 ROM works). Maybe it's normal in current state?

command line : m65 -l com6 -F -r test.prg

[image]

Taygair, 2021-08-19:
Might me similar to this one: https://github.com/MEGA65/mega65-tools/issues/2

gardners, 2021-08-19:
Works fine for me from under Linux.
Is it a Windows-specific problem perhaps?
[image]

(Note that the for the image above it was a C64 program I loaded, thus why it didn't run. But you can see that there was no FILE NOT FOUND error)

Are you using the latest m65 binary?

Taygair, 2021-08-19:
Tested on Mac to real hardware (R2), works also here:
- 920217.BIN
- mega65r2-dev.bit (V116)
-> Loaded with alpaca.prg

Majikeyric, 2021-08-19:
I use Windows 7.
M65 version : 20210525.08-HEAD_n-713db01
bitstream : 116-0fb485d

Majikeyric, 2021-08-19:
I updated M65 to the latest version (20210727) and I have the same issue.

Majikeyric, 2021-08-23:
With latest BS ROM (920219), it works again.

Taygair, 2021-08-23:
Excellent and thank you Paul!

closed 2021-08-23 by Taygair