MEGA65/mega65-tools #108: m65 -q quiet switch

jesperrula, 2022-01-24:
I use the m65 tool as part of my toolchain for cross platform development. It would be nice if the m65 tool had a -q switch that would reduce the stdout to just output filename, load start and end address. 

Example output could be: 
Loading '%filename' to $2001-$3456

If there are errors, these should be piped to stdout.

lydon42, 2022-01-30:
The proper way to do this is to implement proper logging with different levels of verbosity. Levels could be: normal, verbose, debug. In addition a quiet level could reduce this to only critical error messages.
This means: go through the whole thing and find all logging messages, categorizing, adding switches for verbosity.

lydon42, 2022-06-26:
Goal is to reduce the clutter of debug messages.

lydon42, 2022-07-24:
proper logging has now been implemented.
Option to control is -0 (dash zero) as all letters are already taken. Parameter takes number from 0 to 5 or a word describing the severity (crit, erro, warn, note, info, debu).

lydon42, 2022-09-25:
Waiting to be merged to master.

closed 2022-09-25 by lydon42