MEGA65/mega65-tools #111: (WINDOWS) mega65_ftp adds invalid root directory changes to the current directory path

mindrail, 2022-05-18:
## When you change to an invalid directory by prepending / it changes you to an invalid directory.

### Examples below:

MEGA65 SD-Card:/> cd /hello
ERROR: Could not find directory segment 'hello'
ERROR: Could not open directory '/hello'
MEGA65 SD-Card:/hello> dir
ERROR: Could not find directory segment 'hello'
0 Dir(s), 0 File(s)
MEGA65 SD-Card:/hello> cd /should/not/change/path/if/the/path/doesnt/exist
ERROR: Could not find directory segment 'should'
ERROR: Could not open directory '/should/not/change/path/if/the/path/doesnt/exist'
MEGA65 SD-Card:/should/not/change/path/if/the/path/doesnt/exist> dir
ERROR: Could not find directory segment 'should'
0 Dir(s), 0 File(s)
MEGA65 SD-Card:/should/not/change/path/if/the/path/doesnt/exist> cd /
MEGA65 SD-Card:/> dir *.ROM
      131072 | 2022-05-12 02:07:40  | MEGA65.ROM   |
      131072 | 2022-05-06 20:39:56  | MEGA651.ROM  |
      131072 | 2022-05-06 20:39:56  | MEGA652.ROM  |
      131072 | 2022-05-12 02:06:40  | MEGA653.ROM  |
0 Dir(s), 4 File(s)
MEGA65 SD-Card:/>

## Behavior should mirror what happens if you normally try to change to an invalid directory:

MEGA65 SD-Card:/> cd poo
ERROR: Could not find directory segment 'poo'
ERROR: Could not open directory '/poo'
MEGA65 SD-Card:/>

lydon42, 2022-08-01:
happens only with leading slash

lydon42, 2022-08-01:
should be fixed, please check with latest version

closed 2022-08-01 by lydon42