MEGA65/mega65-tools #117: m65 causes bad MEGA65 state

johanberntsson, 2022-06-03:
I'm trying mounting and running a d81 form the script using
m65 -d mygame.d81 -1 autoboot.c65
The program doesn't start and the MEGA goes into some strange state where I need to power cycle (reset doesn't work properly).
Perhaps some of the workload within the m65 tool tries to pause the cpu, and un-pause it at certain times, and that logic has been muddled up due to some recent change done to m65 tool.

lydon42, 2022-07-25:
So, I am not sure here...
You do mount a D81 virtually via m65. And then you tell it to load autoboot.c65 from your PCs disk to the MEGA65.
Is this what you intend to do?

lydon42, 2022-07-25:
Just checked with `m65 -d SOLI65.D81 -1 ~/Downloads/bsram.prg`:
System has access to the D81 and the bsram.prg is loaded into memory and can be started. All looks good.
With autoboot.c65 as argument m65 failes, as I have no autoboot.c65 on my PCs harddrive.
So I would say this is no bug, and there is no unstable situation.

Please supply exact versions used.

lydon42, 2022-08-01:
I guess you want the your command be something like:
`m65 -d mygame.d81 -T 'boot:'`
just checked this with the GEOS65.D81, Works for me in both master and merge2-m65dbg. The later is recommended on unix systems as it uses select to cut the 100% CPU usage.

closed 2022-08-01 by lydon42