MEGA65/mega65-tools #121: Segfault when using dput on a long or complicated filename

faxm0dem, 2022-06-25:
## Test env

* Linux 5.10.100 1 SMP Tue Feb 15 13:56:02 CET 2022 x86_64 GNU/Linux
* tools version 0ece8303b600997f22aaebdf0a77103a870bc041

## Bug description

When using `mega65_ftp` on a `sdcard` and `dput` using a file with special chars yields a segmentation fault.

## Reproduction

```
# sudo bin/mega65_ftp -d /dev/mmcblk0
[...]
MEGA65 SD-Card:/c64> ldir *.PRG
       46231 INT KARATE +_VGG.PRG
MEGA65 SD-Card:/c64> dput INT KARATE +_VGG.PRG
Segmentation fault (Core dumped)
```

## Stack Trace

```
(gdb) bt full
#0  0x00007f04998f00c1 in __GI__IO_fread (buf=0x7fff74351202, size=1, count=254, fp=0x0)
    at iofread.c:37
        _IO_acquire_lock_file = <optimized out>
        bytes_requested = 254
        bytes_read = <optimized out>
#1  0x00005598198c4046 in add_prg (fname=0x7fff743518a0 "INT") at src/tools/diskman.c:180
        f = 0x0
        prgname = "INT\000\377\177\000\000\250\023\065t\377\177\000\000\000\000\000\000\000\000\000\000\250\023\065t\377\177\000 \310{\327\032\230U\000\000\305{\327\032\003\000\000\000p\023\065t\377\177\000\000uP\213\031\230U\000\000\234\023\065t\377\177\000\000\260\023\065t\377\177\000\000\250\023\065t\377\177\000\000\240\023\065t\377\177\000\000\250\023\065t\377\177\000\000\240\023\065t\001\000\000\000\200\024\065t\377\177\000\000\222R\213\031\230U\000\000\b\223\217\031\230U\000\000\300{\327\032\230U\000\000\000\000\000\000\000\000\000\000`\265\243\231\001\000\000\000\017\223\217\031\230U\000\000\310{\327\032\230U\000\000\030\000\000\000\060\000\000\000\377\377\377\377\377\000\000\000\002\000\000\000(\000\000\000\000"...
        chunk = '\000' <repeats 255 times>
        starttrack = 1
        startsector = 0
        curtrack = 1
        cursector = 0
        prevtrack = 0
        prevsector = 0
        bytes = 32767
        sectorcnt = 0
#2  0x00005598198c410a in create_d81_for_prg (prgfname=0x7fff743518a0 "INT")
    at src/tools/diskman.c:215
        d81name = '\000' <repeats 255 times>
#3  0x00005598198ba943 in wrap_upload (fname=0x7fff743518a0 "INT")
    at src/tools/mega65_ftp.c:2810
        d81name = 0x1 <error: Cannot access memory at address 0x1>
#4  0x00005598198b552c in execute_command (cmd=0x55981ad77bc0 "dput INT KARATE +_VGG.PRG ")
    at src/tools/mega65_ftp.c:486
        cluster_num = 0
        sector_num = 32767
        slot = 0
        src = "INT", '\000' <repeats 85 times>, "z@\220\231\004\177\000\000\000\000\000\000\000\000\000\000\340\233\243\231\004\177\000\000\000\000\000\000\000\000\000\000\314A\220\231\004\177\000\000 \000\000\000\000\000\000\000\001\000\000\000\000\000\000\000\002", '\000' <repeats 31 times>, "\002\000\000\000\060\000\000\000@?\220\231\004\177", '\000' <repeats 18 times>...
        dst = '\000' <repeats 1023 times>
#5  0x00005598198b629f in main (argc=3, argv=0x7fff743525f8) at src/tools/mega65_ftp.c:825
        cmd = 0x55981ad77bc0 "dput INT KARATE +_VGG.PRG "
        ret = 18
        opt = -1
        prompt = "MEGA65 SD-Card:/> \000\231\004\177\000\000P?\247\231\004\177\000\000@\"5t\377\177\000\000\a\000\000\000\000\000\000\000\a\000\000\000\b\000\000\000\360\064\247\231\004\177\000\000\350驙\004\177\000\000M\362\247\231\004\177\000\000\000\000\000\000\000\000\000\000\330\000\250\231\004\177\000\000\220!5t\377\177\000\000\300z\211\231\004\177\000\000\b", '\000' <repeats 15 times>, "\020\"5t\377\177\000\000\000\000\000\000\000\000\000\000\020\"5t\377\177", '\000' <repeats 18 times>, "`7\247\231\004\177\000\000\350\364\251\231\004\177\000\000\204\071\247\231\004\177\000\000\060\064\247\231\004\177\000\000Hਖ਼"...
```

gurcei, 2022-06-25:
Haven't had a chance to try replicating this issue yet, but just as a workaround for now, will it work if you surround the long filename with double quotes?

faxm0dem, 2022-06-28:
I'll try that ASAP

faxm0dem, 2022-06-30:
That worked
```
MEGA65 SD-Card:/> dput "INT KARATE +_VGG.PRG"
Wrapping "INT KARATE +_VGG.PRG" into "INT KARATE +_VGG.D81"...
INT KARATE +_VGG.D81 does not yet exist on the file system -- searching for empty directory slot to create it in.
- Using DOS8.3 name of 'INTKAR~1.D81'
Uploaded 819200 bytes in 1 seconds (800.0KB/sec)
MEGA65 SD-Card:/>
```

faxm0dem, 2022-06-30:
well, it's still a bug, maybe not close it so fast ?

lydon42, 2022-06-30:
TODO:
- don't segfault on weird chars, but warn user to use normal chars and 8.3 names.
- implement proper LFN handling

gurcei, 2022-07-02:
Hmm, I feel like the usage of quotes to surround filenames containing spaces is standard practice in a lot of OS. Not doing so usually leads to "un-expected" behaviour.

Ok, in this case though, getting a segfault is a surprise and worth sorting out, so that it handles the situation a bit more gracefully :-) I'll try look into when I get a chance.

lydon42, 2022-07-25:
segfault averted!

closed 2022-07-25 by lydon42