MEGA65/mega65-tools #129: PRG send causes BS Monitor's X command to fail

dansanderson, 2022-07-29:
**Test Environment (required)**
 - Tools Platform: MacOS
 - Tools Version: 20220519.21-HEAD_n-d9a2452
 - MEGA65 Platform: MEGA65
 - ROM Release: 920376
 - Core Commit: f555316

**Describe the bug**
Something about the way m65 sends PRG files confuses BS Monitor's `X` command. The `X` command is supposed to exit to BASIC, but if the program was sent via m65/M65Connect, `X` actually goes somewhere else then re-breaks back into the monitor with PC `002005` (which is a weird PC because that's in BASIC memory).

**To Reproduce**
1. Transfer any PRG from a PC to the MEGA65. Don't run it or switch modes. `m65 -l /dev/cu... somefile.prg`
2. On the MEGA65 at the `READY.` prompt, type `MONITOR`. BS Monitor opens.
3. Type `X`, hit Return.

**Expected behavior**
The `X` command is supposed to exit the monitor and return to BASIC. It works if no program is loaded, or if a program is loaded from disk/SD card. It doesn't matter if the monitor was invoked by the `MONITOR` command or a `brk` instruction from the program.

Instead, the monitor attempts to exit, but then re-breaks with the PC at 002005. This appears to be true only if the PRG was transferred by m65/M65Connect. The content of the program does not appear to matter, and you don't have to run the program to reproduce the issue.

**Additional context**
Here's the beginning of `monexit`, which is where `X` goes:

```
cd9b          monexit
cd9b 20 80 03  jsr restore_system ;Restore system configuration
cd9e ad 05 11  lda save_status  ;Restore BASIC's init status
cda1 8d 04 11  sta init_status
cda4 ea        nop  
cda5 6c 00 11  jmp (system_vector) ;Return to system
```

(Issue relocated from https://github.com/MEGA65/m65connect/issues/11.)

dansanderson, 2022-07-29:
[image]

dansanderson, 2022-07-29:
Trying with the refactor-ftp branch, `20220729.00-refact-00e36ae`. Can still repro.

```
% ./bin/m65.osx --debug -l /dev/cu.usbserial-251633006E0EA ~/websites/site-dansanderson.com/content/lab-notes/2022/07/26/mega65-monitor/golml.prg
2022-07-29T07:36:06.785Z DEBG parameter parsing done
2022-07-29T07:36:06.786Z NOTE MEGA65 Cross-Development Tool 20220729.00-refact-00e36ae
2022-07-29T07:36:06.786Z INFO opening serial port /dev/cu.usbserial-251633006E0EA
2022-07-29T07:36:06.788Z DEBG set_serial_speed: 2000000 bps (OSX)
2022-07-29T07:36:06.935Z INFO checking if MEGA65 has RX buffer
2022-07-29T07:36:06.959Z INFO RX buffer detected. Latency will be reduced.
2022-07-29T07:36:06.979Z INFO detecting C64/C65 mode status
2022-07-29T07:36:07.075Z DEBG in C65 mode
2022-07-29T07:36:07.075Z NOTE loading file '/Users/dan/websites/site-dansanderson.com/content/lab-notes/2022/07/26/mega65-monitor/golml.prg'
2022-07-29T07:36:07.091Z INFO LOAD vector from ROM is $f4c4
2022-07-29T07:36:07.139Z DEBG Injecting string into key buffer at $02B0 : DLo"![$0d]
2022-07-29T07:36:07.139Z INFO waiting for breakpoint at $F4C4 to trigger
2022-07-29T07:36:07.155Z INFO breakpoint @ $F4C4 triggered
2022-07-29T07:36:07.235Z INFO requested file is '0:!' (len=3)
2022-07-29T07:36:07.235Z INFO forcing load address to $2001
2022-07-29T07:36:07.235Z DEBG read block for $2001 -- $2290 (656 bytes)
2022-07-29T07:36:07.363Z INFO storing end of program pointer at $82
2022-07-29T07:36:07.363Z INFO returning top of load address = $2291
2022-07-29T07:36:07.411Z NOTE loaded
```

dansanderson, 2022-07-29:
It appears some bytes at `$000380` are getting corrupted by the transfer, specifically:
```
 . 0380  A2 91    LDX  #$91
 . 0382  A0 22    LDY  #$22
 . 0384  18       CLC
 . 0385  60       RTS
```

These locations need to be preserved for a clean exit from the monitor. Original code before it is overwritten:
```
 . 0380  AD 10 01 LDA  $0110
 . 0383  AE 11 01 LDX  $0111
```

lydon42, 2022-07-30:
should be fixed in refactor-m65

lydon42, 2022-08-01:
also fixed in merge2-m65dbg

dansanderson, 2022-08-01:
The latest in merge2-m65dbg resolves this issue. Per @lydon42 I'll close this ticket. Thanks very much!

closed 2022-08-01 by dansanderson