MEGA65/mega65-tools #145: Successful command reports serial port failure

dansanderson, 2022-09-22:
**Test Environment (required)**
 - Tools Platform: MacOS
 - Tools Version: 20220922.10-refact-8bbd38e
 - MEGA65 Platform: MEGA65
 - ROM Release: 920377
 - Core Commit: 9c03f8a

**Describe the bug**
`m65 -l ... -b ...` successfully sends a bitstream, but then reports a critical failure finding the serial port.

This is the [latest refactor_m65 version build](https://github.com/MEGA65/mega65-tools/releases/tag/CI-refactor-latest) for macOS, running on an Intel Mac macOS 12.6. (Set-up procedure: `mv m65.osx m65`, `chmod 755 m65`, then right-click in Finder to Open without arguments to authorize its use. Then it runs from the command line.)

**To Reproduce**
```
% ~/Downloads/m65 -l /dev/cu.usbserial-251633006E0EA -b ~/Downloads/mega65r3-reltest\ 2/mega65r3-20220922.16-develo-9c03f8a.bit
2022-09-22T22:56:25.840Z NOTE MEGA65 Cross-Development Tool 20220922.10-refact-8bbd38e
2022-09-22T22:56:25.868Z NOTE selecting device UNKNOWN (Digilent; 0403:6010; 251633006E0E; 03636093)
2022-09-22T22:56:25.987Z NOTE fpgajtag: Starting to send file
2022-09-22T22:56:29.589Z NOTE fpgajtag: Done sending file
2022-09-22T22:56:29.608Z NOTE Bitstream loaded
2022-09-22T22:56:29.608Z NOTE no serial device given, trying brute-force autodetect
2022-09-22T22:56:29.608Z NOTE tip: try using 'm65 --autodiscover --verbose'
2022-09-22T22:56:29.608Z CRIT serial port not specified, aborting.
```

lydon42, 2022-09-23:
We need debug output to understand why auto-detection did not work for you.

dansanderson, 2022-09-23:
Notice that I provide a USB device with `-l`. It should not be auto-detecting at all, let alone reporting that auto-detection failed.

My MEGA65 is dead for the moment, but I assume the debug output is useful even when it's turned off, so:

```
% ~/Downloads/m65 --debug --autodiscover
2022-09-23T06:19:41.028Z DEBG parameter parsing done
2022-09-23T06:19:41.029Z NOTE MEGA65 Cross-Development Tool 20220922.10-refact-8bbd38e
2022-09-23T06:19:41.029Z INFO using default FPGA ID ffffffff
2022-09-23T06:19:41.055Z INFO found 1 candidate USB devices.
2022-09-23T06:19:41.055Z INFO found Digilent (serial 251633006E0E) [20, [4]]
2022-09-23T06:19:41.059Z INFO   got 16 id codes: 0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff,0fffffff
2022-09-23T06:19:41.059Z INFO   device seems to be disabled, please power on
2022-09-23T06:19:41.059Z INFO requested id ffffffff does not match any usb interface
2022-09-23T06:19:41.059Z CRIT no valid serial port not found, aborting
```

dansanderson, 2022-09-23:
I stand corrected, I can turn on my MEGA65 without a core in slot 0 and get a better result:

```
% ~/Downloads/m65 --debug --autodiscover
2022-09-23T06:26:21.339Z DEBG parameter parsing done
2022-09-23T06:26:21.339Z NOTE MEGA65 Cross-Development Tool 20220922.10-refact-8bbd38e
2022-09-23T06:26:21.339Z INFO using default FPGA ID ffffffff
2022-09-23T06:26:21.359Z INFO found 1 candidate USB devices.
2022-09-23T06:26:21.359Z INFO found Digilent (serial 251633006E0E) [20, [4]]
2022-09-23T06:26:21.363Z INFO   got 1 id code: 03636093
2022-09-23T06:26:21.363Z INFO   id code 03636093 matches ffffffff
2022-09-23T06:26:21.363Z NOTE #0: UNKNOWN (Digilent; 0403:6010; 251633006E0E; 03636093)
2022-09-23T06:26:21.363Z NOTE selecting device UNKNOWN (Digilent; 0403:6010; 251633006E0E; 03636093)
```

ki-bo, 2022-09-24:
The change I added helps to select the correct usbserial device. The former check (selecting the device ending with a '1') was too simple and did not work on that specific setup (note that in the output shown above no usbserial device is claimed to be found).

ki-bo, 2022-09-27:
Is this issue now confirmed to be solved for you with the recent build of m65.osx?

dansanderson, 2022-09-27:
@ki-bo Confirmed, the latest CI build from the refactor_m65 branch including your chagnes works on my machine. Specifying `-l` does not result in spurious auto-discovery messages, and auto-discovery itself appears to be working well (no `-l` needed).

I assume we're resolving issues after committing fixes to the refactor_m65 branch, so I'll go ahead and resolve this one. Reopen if needed to track merges.

Thanks very much!

closed 2022-09-27 by dansanderson