MEGA65/mega65-tools #156: m65 PRG injection load address is undocumented and misleading

dansanderson, 2023-01-02:
**Test Environment (required)**
 - Tools Platform: MacOS
 - Tools Version: 20221008.12-refact-093eae9
 - MEGA65 Platform: MEGA65
 - ROM Release: 920377
 - Core Commit: 2f545d1

**Describe the bug**
When I run `m65 myfile.prg`, it [reads the load address](https://github.com/MEGA65/mega65-tools/blob/master/src/tools/m65.c#L2631), then [types](https://github.com/MEGA65/mega65-tools/blob/master/src/tools/m65.c#L2635) on the MEGA65 `REM M65 INJECTING AT {load_addr}`. It then tests the load address to see [if it's a SID file](https://github.com/MEGA65/mega65-tools/blob/master/src/tools/m65.c#L2639), and if not, it [overrides the load address](https://github.com/MEGA65/mega65-tools/blob/master/src/tools/m65.c#L2746) based on command-line switches. If my PRG file starts at $3000, m65 announces that it is loading at $3000, but actually loads at $2001.

There is a command line argument, `-1`, that stops this override and honors the load address. However, this argument is not documented under `-h` help (at least in the version I'm using).

**Fix 1: Report the injection address accurately**
Regardless of other changes, m65 should not misreport the injection address. `REM M65 INJECTING AT {load_addr}` should state the actual injection address whether it is derived, overridden, or specified on the command line.

**Fix 2: Document the -1 flag**
If the `-1` flag is to be the way the user tells m65 to honor the load address of the PRG, it should be documented in the `-h` help text.

**Fix 3: Make -1 the default behavior**
I can't think of a reason why m65 should override the load address unless I specifically tell it to with the `-@` flag. As written, m65 assumes that any PRG that doesn't have one of two SID load addresses is either a C64 BASIC PRG if the `-4` argument is provided or a C65 BASIC PRG if the `-1` argument is *not* provided. An actual BASIC PRG should have the correct load address in its header. I believe [this block](https://github.com/MEGA65/mega65-tools/blob/master/src/tools/m65.c#L2746) should be removed.

If there's a use case I'm missing and `-@` doesn't meet that case, I would expect "force a BASIC load address" to be a separate flag that defaults to off and decides which address to use from the presence of the `-4` flag.

lydon42, 2023-01-02:
I agree that -1 should be in the help and that the printed text should match the address. 

Everything else changes the behavior of the switches, which we are trying to avoid.

lydon42, 2023-01-02:
One additional problem: 
You can use a filename and -@ at the same time. But -@ will use overwrite the filename specified for the normal load, so you can't really use both. This is an artifact from the original confusing flow of growing m65 tool options.
I think it would be nice if both work concurrently... so this is also an enhancement.

closed 2023-01-02 by lydon42