MEGA65/mega65-tools #164: mega65_ftp

Taygair, 2023-04-23:
With the new version of M65Connect I want to support long filenames. I've experienced issues with some commands:

**MOUNT** a D81 image with a space in filename or bigger than 8 chars: Will be processed with no error message but image will not be mounted
Example: mega65_ftp.osx -l /dev/cu.usbserial-25163305977F1 -s 2000000 -c "mount \"MY REAL ORDER.D81\""

**RENAME** doesn't seem to consider case sensitivity:
mega65_ftp.osx" -l /dev/cu.usbserial-25163305977F1 -s 2000000 -c "rename \"My File.D81\" \"My FILE.D81\""

Output: _ERROR: Cannot rename to "My FILE.D81", as this file already exists._

gurcei, 2023-04-25:
Will just share some thoughts on how double-quotes within double-quotes, tends to be tricky (and probably isn't related to all the matters you mentioned, but maybe some).

The way it can be handled can vary depending on whether you're trying to get it to work:
- from the console (which also varies depending on OS)
- from within your code (which could vary depending on language choice: c/xojo/etc)

# Getting double-quotes-within-double-quotes to work in the console

## Linux/Mac osx

### method 1: escaping internal double-quotes

With this technique, you could just escape the internal double-quotes with the backslash character.

`mega65_ftp.odx -l /dev/cu... -s 2000000 -c "rename \"My File.D81\" \"MY FILE.D81\" "`
- Note all internal double-quotes now have backslashes
  - `\"My File.D81\"
  - `\"MY FILES.D81\"`
- Note that the external double-quotes don't need the backslashes
  - `-c "rename...`
  - `...D81\" "`

### method 2: avoidance: use single-quote and double-quote

Another technique is to avoid the situation by using single-quotes externally, but double-quotes internally. E.g.:

`mega65_ftp.odx -l /dev/cu... -s 2000000 -c 'rename "My File.D81" "MY FILE.D81" '`

- note the single-quote used for the `'rename` (instead of double-quote)
- note the single-quote used for the end `.D81" '` (instead of double-quote)

(I added a space between the last double-quote and last single-quote just for clarity, the extra space isn't really needed)

Taygair, 2023-04-25:
Thanks for the explanation! I'm aware of it and remembered the discussion about it we had in in the past. In between the mega65_ftp was enhanced, partly restructured/refactored so I gave it another try to support longer filenames with spaces and upper/lowercases. That all works fine on rename (well there is sometimes a weird behavior that I can't reproduce), delete and also copy.
For removing visual complexity I wanted to show some simple examples and removed what I REALLY send, what made you thinking I'm not escaping double quotes So in truth I'm working with \" on mount, rename, copy and delete.

Concrete example of what I'm sending for a mount command:
`mega65_ftp.osx -l /dev/cu.usbserial-25163305977F1 -s 2000000 -c "mount \"My File.D81\""`
That should cover your concerns. Maybe that problem only exists in the OSX versions. Does it work that way on Linux, image got truly mounted?