MEGA65/mega65-tools #167: Quitting mega65_ftp in relation with entering freezer "unmounts" default disk image

Schwefelholz, 2023-07-01:
**Test Environment (required)**
 - Tools Platform: Linux
 - Tools Version: mega65_ftp, 20230701.14-develo-724bf19
 - MEGA65 Platform: MEGA65
 - ROM Release: 920377
 - Core Commit: 23b2368

**Describe the bug**
After quitting mega65_ftp the M65 gets rebooted and mounts the default disk image. `dir` shows the expected content.
After entering the freezer, the disk image is shown not mounted. Leaving the freezer, `dir` shows `74,DRIVE NOT READY,09,37`.

**To Reproduce**
Steps to reproduce the behavior:
1. start mega65_ftp
2. exit mega65_ftp
3. `dir`
4. enter freezer
5. note there is no disk image name shown in (0)
6. leave freezer
7. `dir`
8. see error message

**Expected behavior**
The default disk image should remain mounted.

ki-bo, 2023-08-29:
The d81 seems to be already "unmounted" when the mega65_ftp helper application is transferred into memory.

Schwefelholz, 2023-08-29:
I don't think so. In this case,the `dir` in step 3 shouldn't be working, should it?

lydon42, 2023-08-29:
@Schwefelholz could you please also add versions of mega65-tools and freezer used?

ki-bo, 2023-08-29:
> I don't think so. In this case,the `dir` in step 3 shouldn't be working, should it?

I believe we run into a kind of "semi" mounted state. While the image can still be accessed, hyppo already thinks nothing is mounted. ETHLOAD.M65 is started by hyppo as a "blank" task. I haven't looked into how this works (it was already implemented when I started), but maybe clearing all the pagetables and stuff is erasing hyppo's memory what it mounted.

ki-bo, 2023-08-29:
Right now, not the default disk image is available after quit, but rather the last one mounted before mega65_ftp was started (if any). This is the same behavior as with the uart version of mega65_ftp. As the mounted image is probably still available in lower memory, I could read that string and explicitly call the hyppo mount services. This would make it show up again in freezer.

Schwefelholz, 2023-08-29:
That would make sense, wouldn't it. I'd absolutely expect the last mounted image available after quitting mega65_ftp.

ki-bo, 2023-08-31:
In fact, the image still is mounted by the F011 controller, but the ethernet "hyperrupt" triggers a routine in hypervisor to clear all process descriptor memory. One of the strings in there is the filename of the last mounted image. This is purely cosmetic up to the point where freezer comes into play. It will restore the mounted image by filename when unfreezing (which by the way is also failing if the file was mounted from a sub-directory, as that information is never remembered). If the filename is empty, it will just unmount. That is the reason for this bug.

In order to fix this, we should not erase the complete process control block to keep the mounted image information there. Freezer will see that and try to re-mount that image when un-freezing. The change needs to be done in hyppo, thus in the mega65-core repo.

closed 2023-12-23 by ki-bo