MEGA65/mega65-tools #168: Build of vcdgraph fails on Linux: undefined reference to symbol 'png_write_image@@PNG16_0'

bjotos, 2023-08-15:
**Test Environment (required)**
 - Tools Platform: Linux, build from Source

**Describe the bug**
Check-out the latest version of mega65-tools (9098dea22ae1c7a0bd19da55d88b15df1ffe646c) and try to build on Ubuntu 22.04 LTS.
Linking vcdgraph fails.

This can be reproduced by running the compile step for vcdgraph on its own:

> gcc -Wall -g -std=gnu99 -lusb-1.0 -L/usr/lib -lpng -mno-sse3 -march=x86-64 -I/usr/include/cairo -g -Wall -o bin/vcdgraph src/tools/vcdgraph.c -lcairo
> src/tools/vcdgraph.c: In function ‘main’:
> src/tools/vcdgraph.c:469:13: warning: unused variable ‘v’ [-Wunused-variable]
>   469 |       float v = 0.615*r - 0.515*g - 0.100*b;
>       |             ^
> src/tools/vcdgraph.c:468:13: warning: unused variable ‘u’ [-Wunused-variable]
>   468 |       float u = -.147*r - 0.289*g - 0.436*b;
>       |             ^
> /usr/bin/ld: /tmp/ccRz40FS.o: undefined reference to symbol 'png_write_image@@PNG16_0'
> /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/libpng.so: error adding symbols: DSO missing from command line
> collect2: error: ld returned 1 exit status

**Solution**
The default linker on Ubuntu resolves symbols/dependencies in order of appearance.
The vcdgraph.c introduces a new (and thus unresolved) symbol dependency to png_write_image,
thus the linker needs '-lpng' after this file.

The simple solution is to apply this diff:

> diff --git a/Makefile b/Makefile
> index 1ebf38f..c46aa55 100644
> --- a/Makefile
> +++ b/Makefile
> @@ -642,7 +642,7 @@ $(BINDIR)/bitinfo:  $(TOOLDIR)/bitinfo.c Makefile
>         $(CC) $(COPT) -g -Wall -o $(BINDIR)/bitinfo $(TOOLDIR)/bitinfo.c
>  
>  $(BINDIR)/vcdgraph:    $(TOOLDIR)/vcdgraph.c Makefile
> -       $(CC) $(COPT) -I/usr/include/cairo -g -Wall -o $(BINDIR)/vcdgraph $(TOOLDIR)/vcdgraph.c -lcairo
> +       $(CC) $(COPT) -I/usr/include/cairo -g -Wall -o $(BINDIR)/vcdgraph $(TOOLDIR)/vcdgraph.c -lcairo -lpng
>  
>  # Create targets for binary (linux), binary.exe (mingw), and binary.osx (osx) easily, minimising repetition
>  # arg1 = target name (without .exe)

Note: 
It looks like some other targets (i.e. readdisk) had similar issues and did same trick.
Cleaner (and common) solution would be to differentiate between CFLAGS (i.e. compiler flags) and LFLAGS (i.e. linker flags) and add linker flags at the very end of each build step.

bjotos, 2023-08-15:
Okay, github messed-up the diff (preview looked fine, argh), here is the a diff as attachment
[vcdgraph-linux-compile-fix-diff.txt](https://github.com/MEGA65/mega65-tools/files/12352792/vcdgraph-linux-compile-fix-diff.txt)

lydon42, 2023-08-15:
The -lpng seems to just missing, pkg-config does not generate -lpng for me, as it has no `--libs-only-l` flag, so I wonder where you get the -l in the front.

commit ff5b6507 (2023-08-15): fix(vcdgraph): Makefile is missing -lpng, fixes #168
closed 2023-08-15 by lydon42