MEGA65/mega65-tools #171: ethernet version of mega65_ftp not working with open-roms

gurcei, 2023-08-28:
For occasions when there's no `MEGA65.ROM` on the sd-card and we jump into open-roms (if it the user choose to load the open-roms by preference), it would be good it the new ethernet mega65_ftp tool could still function with open roms.

The existing serial-bsed mega65_ftp already works with open-roms, so hopefully we can get ethernet-based mega65_ftp working with it too.

ki-bo, 2024-01-05:
As the MEGA65_FTP helper is a C64 program compiled with cc65, the default etherload loading routine that is used by mega65_ftp will try to reset to C64 mode and start the helper afterwards in that environment. This of course doesn't work with the open rom where no real C64 mode is available. But there is another way of starting the helper, without doing a reset first. This would just load the data to the memory at $801 and then do a direct jump to its starting address. I may need to add some more initialisation code to the helper to setup all I/O correctly but should be possible and then be totally independent of the rom. The good thing is the helper doesn't need any ROM routines and also is not using any IRQs. This will make it easier to fix this.

ki-bo, 2025-03-27:
This would also allow users to upload a ROM file to the SD card after formatting and re-populating it from a CORE where no ROM binary is present. This was one of the few occasions where I still needed to remove my SD card and connect it to my PC/Mac in order to copy the MEGA65.ROM file to it.