MEGA65/mega65-tools #204: Compilation under Arch Linux doesn't find installed libraries

nils-eilers, 2024-09-25:
Hi there, are there some Arch Linux hackers around?

I'm trying to compile the mega65-tools but though I have readline, ncurses and tfinfo libraries installed, the linker complains it cannot find them. Any suggestions?

```
[j@VB mega65-tools]$ LANG=C make all
gcc -Wall -g -std=gnu99 `pkg-config --cflags-only-I --libs-only-L libusb-1.0 libpng` -mno-sse3 -march=x86-64 -D_FILE_OFFSET_BITS=64 -Iinclude `pkg-config --cflags libusb-1.0` -o bin/mega65_ftp src/tools/mega65_ftp.c src/tools/m65common.c src/tools/logging.c src/tools/ftphelper.c src/tools/ftphelper_eth.c src/tools/filehost.c src/tools/diskman.c src/tools/bit2mcs.c src/tools/etherload/etherload_common.c src/tools/etherload/ethlet_dma_load.c src/tools/etherload/ethlet_echo.c src/tools/etherload/ethlet_all_done_basic2.c src/tools/version.c -Wl,-Bstatic -lreadline -lncurses -ltinfo -Wl,-Bdynamic -DINCLUDE_BIT2MCS
src/tools/mega65_ftp.c: In function 'change_dir':
src/tools/mega65_ftp.c:3042:83: warning: 'snprintf' output may be truncated before the last format character [-Wformat-truncation=]
 3042 |       snprintf(&temp_path[strlen(temp_path)], 1024 - strlen(temp_path), "/%s", seg);
      |                                                                             ^

src/tools/mega65_ftp.c:3042:13: note: 'snprintf' output between 2 and 1025 bytes into a destination of size 1024
 3042 |             snprintf(&temp_path[strlen(temp_path)], 1024 - strlen(temp_path), "/%s", seg);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/bin/ld: cannot find -lreadline: No such file or directory
/usr/bin/ld: cannot find -lncurses: No such file or directory
/usr/bin/ld: cannot find -ltinfo: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:839: bin/mega65_ftp] Error 1
[j@VB mega65-tools]$ ls -l /lib/*readline* /lib/*ncurses* /lib/*tinfo*
-rw-r--r-- 1 root root     20 25. Mai 08:43 /lib/libncurses++.so
-rw-r--r-- 1 root root     18 25. Mai 08:43 /lib/libncurses.so
lrwxrwxrwx 1 root root     19 20. Jul 15:48 /lib/libncurses++.so.5 -> libncurses++.so.5.9
lrwxrwxrwx 1 root root     17 20. Jul 15:48 /lib/libncurses.so.5 -> libncurses.so.5.9
-rwxr-xr-x 1 root root  75840 20. Jul 15:48 /lib/libncurses++.so.5.9
-rwxr-xr-x 1 root root 355568 20. Jul 15:48 /lib/libncurses.so.5.9
-rw-r--r-- 1 root root   8448 25. Mai 08:43 /lib/libncurses++w_g.a
-rw-r--r-- 1 root root 185526 25. Mai 08:43 /lib/libncursesw_g.a
lrwxrwxrwx 1 root root     18 25. Mai 08:43 /lib/libncurses++w.so -> libncurses++w.so.6
lrwxrwxrwx 1 root root     16 25. Mai 08:43 /lib/libncursesw.so -> libncursesw.so.6
lrwxrwxrwx 1 root root     20 20. Jul 15:48 /lib/libncurses++w.so.5 -> libncurses++w.so.5.9
lrwxrwxrwx 1 root root     18 20. Jul 15:48 /lib/libncursesw.so.5 -> libncursesw.so.5.9
-rwxr-xr-x 1 root root  75840 20. Jul 15:48 /lib/libncurses++w.so.5.9
-rwxr-xr-x 1 root root 404752 20. Jul 15:48 /lib/libncursesw.so.5.9
lrwxrwxrwx 1 root root     20 25. Mai 08:43 /lib/libncurses++w.so.6 -> libncurses++w.so.6.5
lrwxrwxrwx 1 root root     18 25. Mai 08:43 /lib/libncursesw.so.6 -> libncursesw.so.6.5
-rwxr-xr-x 1 root root  75840 25. Mai 08:43 /lib/libncurses++w.so.6.5
-rwxr-xr-x 1 root root 453896 25. Mai 08:43 /lib/libncursesw.so.6.5
lrwxrwxrwx 1 root root     16  3. Aug 08:18 /lib/libreadline.so -> libreadline.so.8
lrwxrwxrwx 1 root root     18  3. Aug 08:18 /lib/libreadline.so.8 -> libreadline.so.8.2
-rwxr-xr-x 1 root root 343064  3. Aug 08:18 /lib/libreadline.so.8.2
-rw-r--r-- 1 root root     24 25. Mai 08:43 /lib/libtinfo.so
lrwxrwxrwx 1 root root     24 20. Jul 15:48 /lib/libtinfo.so.5 -> /usr/lib/libncurses.so.5
lrwxrwxrwx 1 root root     16 25. Mai 08:43 /lib/libtinfo.so.6 -> libncursesw.so.6
```

lydon42, 2025-01-12:
I exclusively compile under arch. My lib looks the same.

Those files should also be in `/usr/lib`, I think gcc will look there. Please check your setup. You can contact me on discord for more help.

closed 2025-01-12 by lydon42

nils-eilers, 2025-01-12:
Yes, they are in /usr/lib too, but adding  `-Wl,--verbose` to the Makefiles variable COPT showed that the compiler is looking for .a files only, but no .so files. Please don't ask me why that is.

Compilation with `make DO_STATIC=0` finally did the trick, solving the issue.