MEGA65/mega65-tools #217: After Etherload, can't change CPU speed with SPEED command

dansanderson, 2025-03-17:
**Test Environment (required)**
You can use MEGAINFO from within FREEZER to obtain this information
(long press RESTORE to enter FREEZER, then press HELP to start MEGAINFO)
 - Platform: MEGA65R6
 - Core Commit: v0.96 (3c10488)
 - ROM Release: 920412

**Describe the bug**
After any use of Etherload to transfer a file to the SD card, to load a ROM, or load a PRG, the BASIC 65 `SPEED` command cannot be used to change the CPU speed.

I would call this a low priority edge case except it's probably worth digging up an explanation, in case there's a larger issue.

**To Reproduce**
Steps to reproduce the behavior:
1. On PC: `mega65_ftp -e -c put "somefile" -c exit` File transfers, system resets.
2. Open Freezer, confirm CPU speed is 40 MHz.
3. On MEGA65: `SPEED 1`
4. Open Freezer.

Expected: CPU speed is now 1 MHz.
Actual: CPU speed is still 40 MHz.

Repeat with step 1 as `etherload -R mega65.rom`. Same result.

Press Reset button to do a full warm boot with the ROM from the SD card. `SPEED 1` now takes effect as expected.

**Additional context**
I noticed this while testing changes to the `SPEED` command, sending test ROMs via etherload. If that's the only affected use case, I just got unlucky. Ideally any use of etherload shouldn't get the state stuck like this.

ki-bo, 2025-03-18:
mega65_ftp is forcing 40 MHz at the moment. The only way to exit from it is either power-cycle, reset button, using etherload or the "quit" command. All those except for "quit" are also resetting the "force 40 MHz" state in the core. I think it will be enough to just add writing 64 to (CPU port) register 0 right before we reset when the user enters "quit".

ki-bo, 2025-03-18:
By the way, you mentioned that once you exit `mega65_ftp`, uploading and starting a ROM via `etherload -R` doesn’t help. However, that’s not the behavior I’m seeing. After quitting `mega65_ftp`, the `SPEED 1` command does indeed have no effect. But if I then run `etherload -R 920413.bin`, the ROM starts, and `SPEED 1` becomes functional again (as confirmed by the FREEZER’s speed display).

ki-bo, 2025-03-18:
Should be fixed in development HEAD now

dansanderson, 2025-03-27:
Confirmed fixed.

closed 2025-03-27 by ki-bo